juami/pytentiostat #2: git discussion thread

sbillinge, 2019-02-14:
thread for getting help with git issues

sbillinge, 2019-02-14:
@mspence2 typing this will tag you and send you an email

sbillinge, 2019-02-14:
my username @sbillinge

jlhitt1993, 2019-02-14:
@jlhitt1993 Jeremy

mspence2, 2019-02-14:
@mspence2

aplymill7, 2019-02-14:
@aplymill7 Austin

YaoTomTong, 2019-02-14:
@YaoTomTong Yao Tong

sbillinge, 2019-02-14:
call was dropped, but no worries, I think we got everything covered.  I am excited about this project, it should be fun!   post questions here when you have them.  Relevant link, for the record is:
https://github.com/Billingegroup/group_git_workflow

sunhwibang, 2019-02-14:
@sunhwibang Sunhwi

mspence2, 2019-03-07:
I am having trouble with internet, and I keep getting booted from the call. I need to head to class anyway. When I get out of class, I will fetch and merge my master branch and reopen the the authors file to see if it is updated. I also plan to work on the use case that I created by addressing the comments you gave me. Let me know if there are any additional tasks you all discuss.

dimacio, 2019-12-23:
Hello, I don't know where to post this question, so I think to ask it here. I tried to build the JUAMI potentiostat according to the specification in the paper and I can't get a voltammogram using the VC method. I check the signal before the first op-amp it gives a good triangle wave but I get flat signals at the measure point of the circuit. Someone could tell me where to find an explanation of the function of each op-amp? Using the Fig S6 I figure out that the op-amp at the bottom give a reference voltage of 2.5 V and the one at the right is a current follower. I guess that the op-amp connected to the RE avoid the pass of current trough it, But I can't understand the connection of the first op-amp, the one at the upper-left corner. I read others potentiostat circuit published, and they pass the pwm signal trough the non-inverting input of the op-amp.

mspence2, 2020-04-06:
Cleanliness question - when a local branch is pulled into the main repo and is deleted online, how do I remove it from the branch list on my computer (when I type git branch)? 
Should I use git push origin --delete <branchname>?

jlhitt1993, 2020-04-06:
git branch -d branchname.

If it says the branch has not been synced with remote but you know you want
to delete it, use -D

sbillinge, 2020-04-06:
fyi, be careful with -D .  When git doesn't allow you to delete a branch it
is because there are unmerged changes and they will be lost if you -D, so
make sure you really want to blow that thing away before you do that.

Safe workflow is:
```
git checkout master
git pull upstream master
```
then
```
git branch --merged     # it will show branches that are fully merged and
safe to delete
git branch -d <branchname>
```

If you don't sync with upstream it will show a branch as unmerged because,
well, it isn't fully merged!

S