juami/pytentiostat #38: Combining the separate experiments into one code

jlhitt1993, 2019-07-29:
The experiments are divided into LSV, CA, and CV programs. We need to either combine them all or create a main file that will import the individual programs.

sbillinge, 2019-07-29:
in terms of implementation, more modular is always better than less modular, but running from a main is generally preferred.

If we are ambitious we can build a class for running experiments and it can inherit from a base class that handles initalization and setup tasks that are common to all the experiments.

We can also do this using functions.  

The general philosophy is to make functions and class methods as abstract as possible to increase reuse

aplymill7, 2019-07-30:
My preference would be to write a general experiment function in a separate file that will switch between the experiment types based on the experiment type given in the config file. In this file we can have functions for the redundant parts between the experiments. Also, I support doing the initialization outside the main file as well.

I'm not as familiar with classes, but we can probably switch it down the road to a class if it makes it more elegant.

sbillinge, 2019-07-30:
Yes, this sounds like a good plan.

aplymill7, 2019-08-15:
Implemented now

closed 2019-08-15 by aplymill7