juami/pytentiostat #47: UC for operating standard LSV

aplymill7, 2019-07-30:
Fleshed out UC for operating standard LSV for v0.0.2 with dev team audience.

mspence2, 2019-08-04:
@aplymill7 I'm a bit confused about this. Didn't we merge some UCs into upstream in the docs/use-cases folder? Is the plan to restart all of the use cases?

aplymill7, 2019-08-05:
It appears this got lost in a commit reversal. If you look at the page for pytentiostat/docs/use-cases you can  see that the last commit was 19 days ago and was "This reverts commit 83281cd, reversing
changes made to f76e4af." If you click on the link for commit 83281cd it contains yours and my previous UCs. 

We should at least make UCs for a version closer to our working one though as it will making one-to -one testing easier (like if you test something and I want to replicate the error you found to try and troubleshoot).