juami/pytentiostat #56: Identifying the JUAMI potentiostat

jlhitt1993, 2019-08-02:
Currently, the program searches for Arduino Unos that are plugged in and insures that exactly one Uno is plugged in and assumes that this is the potentiostat. It could cause an error if the user has exactly one Uno plugged in that is not a JUAMI potentiostat.

sbillinge, 2019-08-04:
Is it possible for software on the board itself to return a message that it is a JUAMI potentiostat?  This is behavior that we would want.  

Alternatively, what makes our Arduino a JUAMI potentiostat?  Then we can test for that state.  This has the advantage that we can give instructions for people to home-build their own and it would work.

jlhitt1993, 2019-08-04:
It should be possible. The potentiostat will have a unique firmware uploaded to it (StandardFirmata.ino). We can add something to this firmware that will send an identifier but it seems a bit challenging. 

Alternatively, we could do some kind of hardware check. Maybe if an arduino uno is detected, a specific value is written from one pin and read by another to check for the correct value. This would require us to add a wire to the arduinos that wouldn't be in the previously made potentiostats so those could not be used without adding the wire.

jlhitt1993, 2019-08-04:
On second thought, I think I may have an idea for the firmware method. I'll give it a shot.

dimacio, 2019-12-16:
Hi, this is my first experience with git so I'm not used to it yet. I already read https://github.com/Billingegroup/group_git_workflow, but I wasn't sure if I should push a request for this
I had this problem running the main program:

 File "/home/akira/PycharmProjects/pytentiostat/pytentiostat/operator.py", line 137, in read_write
    real_voltage = (pin0value - voltage_offset) * -cf
TypeError: unsupported operand type(s) for -: 'NoneType' and 'float'

I solved it by changing the baud rate of the Standar Firmata protocol on the board from 57600
to 115200 as specified in routine.py.
I think that if we choose a specific baurate for pytentiostat (such as 115200) and verify it when we run _load_arduino () we could solve the problem of identifying the board and the one I had when I wanted to run the main program at the same time. I would like to know what you think of this before continuing to work on this topic.

sbillinge, 2019-12-16:
Thanks for contributing @dimacio .  Do you have one of the potentiostats and are using the code to run the potentiostat?

jlhitt1993, 2019-12-16:
Hello. I think the problem might be that you haven't uploaded the modified version of pyfirmata firmware to an arduino uno board. We don't actually have the firmware in this repository but instead it will be hosted here https://github.com/sbillinge/potentiostat_firmata. It looks like it's not on there at the moment. Maybe @sbillinge can update that.

sbillinge, 2019-12-16:
this is the correct link for the firmata ino:
https://github.com/juami/potentiostat_firmata

(the one Jeremy sent was my fork).  If it needs to be updated we can do
that but I don't have any updated firmware code.

Would it make sense to put instructions how to download and install the
firmware into the README at this github repo?

S

jlhitt1993, 2019-12-16:
Simon, that one looks good for now. 

Yes, instructions for that would be helpful too. I can type something up real quick and send it to you.

dimacio, 2019-12-16:
@sbillinge Thanks to you. I am from Argentina and I think this is a project that can be very useful in Latin America. I made a version of the hardware following the indications of the supplementary information in the original paper and I am testing the software with it. I have experience in electrochemistry and python but I have never worked in collaboration. This is my researchgate account
https://www.researchgate.net/profile/Dimas_Torres
@jlhitt1993 Thanks, that was exactly the problem. I uploaded the correct version to https://github.com/sbillinge/potentiostat_firmata and ran smoothly.

sbillinge, 2019-12-16:
thanks.

S

aplymill7, 2019-12-16:
> Hi, this is my first experience with git so I'm not used to it yet. I already read https://github.com/Billingegroup/group_git_workflow, but I wasn't sure if I should pull a request for this

@dimacio glad things worked out! In the future you can raise this as a new issue. From there I can flag what the issue is and see if it's an external problem or bug with the code. If it turns out it is a bug and you want to resolve it, you can push to a branch on your fork and make a pull request from there. I can test your pull request independently to verify that it fixes the code, nothing else breaks,  and all else is good I can merge to the main code.

dimacio, 2019-12-16:
Thanks @aplymill7,  I currently have a problem to generate the graph from the main application but I will study it before creating a new issue.

aplymill7, 2019-12-16:
Okay great! If it persists to be a problem you can raise a new issue and I will try to help you resolve.

jlhitt1993, 2019-12-16:
@dimacio if you can't fix the problem, let us know about the error you get so we can help you get it fixed.

sbillinge, 2019-12-16:
@dimacio thanks for your interest in this project.  You are welcome to join and make contributions.  The basic steps are:
1) feel free to post issues for the project.  They will be less likely to get lost if you post a new issue for a new, well, issue.
2) fix issues.  Pick an issue to work on and using the group workflow that you already found, post a PR.  In the PR you can say "closes #??"  where ?? is the issue that your PR is fixing.  What happens then is that you will get lots of feedback about your fix, which is a great way to learn programming.  We are very friendly, so feel free to just post a PR and we can work with you to make it pretty and nice and useful.  We are used to that.

We agree strongly that this could be a great thing for Latin America too...

Welcome on board!

Ozturk11, 2021-08-23:
> @dimacio thanks for your interest in this project. You are welcome to join and make contributions. The basic steps are:
>
> 1. feel free to post issues for the project.  They will be less likely to get [...]

We try to make up a potentiostat system for the CV measurements of our supercapacitor system. Then we installed Labview 2019, Arduino Software and JUAMI.exe files on our local computer and created a circuit for electronic components.

But we had a trouble with Software while adjusting voltage average. We choose a value range as an input  (just as; negative 0, positive 1) but it does not exist. We always read the same voltage value 2.5 Volt in the graph.

May you let us solve this problem.

Ozturk11, 2021-08-23:
Dear @sbillinge,
We try to make up a potentiostat system for the CV measurements of our supercapacitor system. Then we installed Labview 2019, Arduino Software and JUAMI.exe files on our local computer and created a circuit for electronic components.

But we had a trouble with Software while adjusting voltage average. We choose a value range as an input  (just as; negative 0, positive 1) but it does not exist. We always read the same voltage value 2.5 Volt in the graph.

May you let us solve this problem.

Ozturk11, 2021-08-23:
@sbillinge We enter the potential value, but it does not reflect the range we entered in the chart