juami/pytentiostat #70: firmata.ino file

sbillinge, 2019-08-20:
I am not sure how this is used.  It is a huge c++ file in our code-base but we don't seem to need to compile it, is that right?  Will it have to be compiled on people's computers for the pytentiostat to run?

jlhitt1993, 2019-08-20:
This is the firmware that tells the arduino to listen for the commands
coming from python. It needs to be compiled and uploaded to the arduino
once as long as the arduino is never reset. If someone is making a
potentiostat from scratch, they will need this.

sbillinge, 2019-08-20:
we should think how to handle this.  It would probably be good to separate
it from the python code.  Are the JUAMI potentiostats shipped with this
already on board?  So the user only does this if he/she is building a
potentiostat from an arduino board, for example?  In that case, we may want
to put it into a separate project even because the target audience is very
different.  It constitutes 60% of code that we are shipping right now.

S

aplymill7, 2019-08-20:
Yeah I agree that maybe this should be on a separate project. My impression is that they could be shipped with this already on them.

PR #73 (2019-08-21): moving firmata c++ code to its own repo
closes #70 

Firmata c++ code now moved to ``potentiostat_firmata`` repo on the JUAMI org.  All the pytentiostat team members should have access to it and this can (should) be developed and documented separately. 

For example, documentation about how to load this onto the arduino and build the code can go there, but this is not something we want most users to do so don't want it in this more user-facing place.
closed 2019-08-21 by aplymill7