juami/pytentiostat #88: Add some pauses to make sure the user is ready

jlhitt1993, 2019-10-13:
After the program is started, we should pause before connecting the potentiostat, running the experiment, and saving the data and ask the user to hit enter in order to continue. This will ensure that the program doesn't try to connect or run the experiment as soon as the program starts in case the user isn't ready. Likewise, the user could abort after the experiment is done, but before data is saved.

aplymill7, 2019-10-14:
I don't see the harm in automatically saving data after generated. If data isn't automatically saved after experiment completion I imagine there would be many cases where one forgets to save before running another experiment and local data is lost. I think the added possibility to run without saving, and adding extra dialog to accommodate this functionality would be similar demand on the user to deleting a data file they don't wish to have.

That being said, I do think an interrupt during the experiment would be helpful. If a user sees something wildly off on the real-time plot and needs to adjust before completing experiment, we should have option to interrupt there. But I still think save the data at that point, and if the user doesn't want to have they can delete.

sbillinge, 2019-10-14:
I don't think Jeremy is suggesting to not save data automatically (that is
a no-brainer) but rather making "pytentiostat connection" and "running an
experiment" to different activities.  I would support that, as in principal
we only need to connect once per session and can run multiple experiments?

S

aplymill7, 2019-10-14:
Ah okay, I was just thinking from "the user could abort after the experiment is done, but before data is saved" that was suggested. 

Definitely connecting once to accommodate multiple runs per session will be good enhancement.

sbillinge, 2019-10-14:
At NSLS-II they have logic that an interrupt (e.g., CTL-C is followed by a
choice of "abort" (no save), "save and end" or "resume".)  We could adopt
similar strategy here (maybe the "resume" is hard but any subset of the
above seems sensible.

  Any interrupt needs to have logic that will reset the state of the system
to something logical, for example, removing any voltages or whatever it
is.

S

jlhitt1993, 2019-10-14:
I was thinking of giving the option to abort without saving because I use that function sometimes on my potentiostat or even in something like Microsoft office when you close the program and choose "Don't save". I am suggesting that we ask the user if they want to save after an abort. I personally think that typing "n" when asked to save is less work for the user than deleting unwanted files afterwards, especially if they have a new file name for every "junk file" then it could get hard for them to remember which ones are junk.

aplymill7, 2019-10-14:
I support it in the case of abort. I like Simon's suggestion of giving the offering in parallel if interrupt used. Something like "abort -a, save and end -s, resume -r".

closed 2019-11-25 by aplymill7