juami/pytentiostat #99: Current/voltage values seem off

jlhitt1993, 2019-10-27:
When I do a scan in the positive direction, the current reads negative and vice versa. Hopefully just a sign issue but I'm not sure if the voltage is flipped or the current.

aplymill7, 2019-10-28:
Can you simultaneously measure with a multimeter to confirm? At least can tell if voltage is.

jlhitt1993, 2019-10-28:
Yes, I tested it and the readings I got matched the values saved and graphed. The issue here is that we get a negative current from a positive voltage... Not sure what to make of it.

aplymill7, 2019-10-28:
Okay yeah that's just a sign issue from definition.

mspence2, 2019-10-29:
Just for confirmation, based on the measurements I have done in the past the voltage applied sweeps the direction that it is told. So it should be a sign issue with the current.

closed 2019-11-25 by aplymill7