juami/pytentiostat #129: Refactor GUI code to MVC

aplymill7, 2019-11-27:
We need to convert the GUI portion of the code to follow the MVC pattern explicitly. Currently, the view portion of the code is largely independent of the  model and controller components, but the MC parts are somewhat intertwined.

Plan is to fully separate this components and explicitly define them as MVC and ensure the code appropriately follows the MVC pattern. Due to time constraints on fixing bugs, this issue will be tackled after AMRS, projected to be complete in Jan. 2020.

weiziyuan, 2019-11-27:
During which, we could also add the multithreading for experiment abortion?

aplymill7, 2019-11-27:
Let's make it a new issue as we can tackle these separately.