juami/pytentiostat #146: distribute flash drives to target audience

sbillinge, 2020-04-01:


aplymill7, 2020-04-02:
I think also if we have the software that we want to implement as well and they have an internet connection, we can tell the target audience to download the set of files to a flash drive and then from the flash drive install. This might be easier to accomplish given current circumstances, but could still accomplish the proof-of-concept.

sbillinge, 2020-04-02:
related to this I just learned about conda-pack which seems to do just what we want.....
https://conda.github.io/conda-pack/ 
I'm not sure if Jeremy wants to change the release at this point to go to this, but either for this release or the next one I think this is a good thing to explore

jlhitt1993, 2020-04-02:
I really like the conda-pack and I think I can incorporate it into the install before the deadline. What I've done now is write a script to create a virtual env in conda just for pytentiostat so it won't interfere with other libraries the user might have on their base python but it sounds like pack might actually be a little simpler.