juami/pytentiostat #147: identify targets for alpha release

sbillinge, 2020-04-01:
make a list of people who will test this.  store it somewhere, not in the repo but associated with the milestone so that we can refer back to it later.

jlhitt1993, 2020-04-02:
How should we store this? Like in a google drive? I feel like some kind of cloud storage would be very beneficial to the team.

aplymill7, 2020-04-02:
I have a box account that I could place on and invite team to. On the box folder can have a link to the milestone for 1-way association (from list to milestone). Or a google drive could work too.

sbillinge, 2020-04-02:
Store here in an issue thread that is linked to the milestone.  It could be this thread! When I say not in the repo I just mean we don't want to commit a file into our code-base that will be there for all time for this....

aplymill7, 2020-04-02:
Okay sounds good! 👍

aplymill7, 2020-04-02:
@sbillinge Are closed issues stored permanently, and if so, would that be a problem?

sunhwibang, 2020-04-16:
Now we have 4 testers identified, so this can be closed.