juami/pytentiostat #150: Freeze software and make alpha release tag in git

sbillinge, 2020-04-01:
(local git not on GitHub for this one)

aplymill7, 2020-04-07:
We should decide where to host this one. 

Also is correct command for this something like: git tag -a v0.1.0alpha -m "Alpha Release" 

Then later to release to access if needed do we just checkout tag like a branch?

sbillinge, 2020-04-07:
That will make a snapshot of the current version.  I am not 100% sure how
to do it but we need to make a release on GitHub, which is different.  I
think rever handles all this if it is set up right.  It is probably worth
google around for "github release" and looking into the rever config files
and do some cross-correlation analysis to try and figure out what is going
on.....

I want to learn this too, so will try and keep an eye on it....

S

aplymill7, 2020-04-07:
I think our rever might be already doing this. So when I look under code/releases there is a tab to look at releases/tags and this contains zip files of the code. They appear to be frozen from time of release.

sbillinge, 2020-04-07:
rever should do it.  It would be nice to understand how it all works
tho.... :)

btw, I failed to do a rever release on my windows machine.  There are some
strange interactions with the OS.  We may need a linux/mac person to run
rever and do the release.

S

aplymill7, 2020-04-07:
Ah I see, looking into..