juami/pytentiostat #151: Make sure the workflow that we expect the target aud to follow is clear

sbillinge, 2020-04-01:
installation....what then?

jlhitt1993, 2020-04-02:
I think we need a demo "experiment" for them to try. If they have a potentiostat then they likely also have a resistor and can run a CV but if they don't have a potentiostat then is there anything they can do? I suppose they could open the GUI just to verify that it runs and all the packages were found.

aplymill7, 2020-04-02:
Is it possible that we can have everyone in our list would have a potentiostat? I think it would be possible since for this alpha we are looking to test a small group.

jlhitt1993, 2020-04-16:
Add a test experiment under basic operation using a resistor or simple conductor like aluminum foil and tell them the expected outcome. Somehow point the alpha testers to these instructions, maybe in the email.