juami/pytentiostat #152: set up ways to formalize the capture of feedback

sbillinge, 2020-04-01:
...to assess the success of the release

jlhitt1993, 2020-04-02:
How about a google forum with a survey? That way we can refer back to the answers and reuse it in the future.

aplymill7, 2020-04-02:
I'm fine with this. Let's determine how we are going to best capture feedback from our survey in today's meeting.

jlhitt1993, 2020-04-02:
So I'm working on a google form to track users feedback. Does anyone have any particular questions they want to see asked? If not, I'll just make a form and send the link to everyone for feedback on the form. I guess recursion is the fastest process for this :)

aplymill7, 2020-04-02:
I think one section could be background : education level, field, experience with hardware, experience with software, experience with command line, experience with python. Second section could be up to basic operation experience. What were weird parts during instillation? what were weird parts during basic operation? Is there some part of instillation/basic operation that you would want improved.

sbillinge, 2020-04-02:
another way to get feedback, if they are supposed to follow a particular
experiment, is to get them to upload the result somewhere, how long and/or
how many attempts it took them to get it, things like that.  A bit of
quantitative data is always good.