juami/pytentiostat #158: Incorporate building docs html into the package installation

mspence2, 2020-04-06:
The goal of this issue is to include a sphinx build process into either the setup.py file or the conda package so that the docs html file will be build upon installation instead of loading build files into the main repo.

mspence2, 2020-04-06:
Copying Simon's latest comments on this:

"We can set things up as we want in terms of build.  It is not normal to put
things that are built from a script into the repo, because a simple build
can make a huge number of files and git will track, line-by-line, all the
changes in every version of the built html files, so it makes the git db
large.

Better is to have a release script (we are using rever but we can have it
do what we want) and whenever we make the release the documents build and
are then put somewhere that is somewhat archival.  In principle, the
release script could build the whole conda-pack and everything we want into
a gzip file that can then be downloaded onto flash drives....

S"

sbillinge, 2025-01-18:
This should close after the recent repackaging

sbillinge, 2025-01-18:
closed by #188.  Now handled by GH release workflow

closed 2025-01-18 by sbillinge