juami/pytentiostat #159: Clean docs folder of outdated and unused files

mspence2, 2020-04-06:
There are a few files that were included in an earlier PR that should have been ignored. They are now outdated and need to be removed from the main repo to avoid confusion.

sbillinge, 2020-04-06:
you can use ``git rm <filename>`` to remove them from your local filesystem but also from the git repo.

mspence2, 2020-04-06:
Sounds good, I was just going to wait until the experiment setup PR is merged, unless you think I should just take care of it on that branch.

aplymill7, 2020-04-07:
@mspence2 I think it's fine to do on current working PR since small change. Just make sure commit note is clear and add to description closes this issue.

mspence2, 2020-04-07:
@aplymill7 for some reason the close statement didn't link this issue to that PR. Maybe you need to manually do it, or we can manually close it. I figure it is best to try to link them for easier tracking if ever necessary.

closed 2020-04-07 by aplymill7