juami/pytentiostat #168: Need to update rever to release to conda-forge and PyPI

jlhitt1993, 2021-04-14:
Looks like the version of pytentiostat on these repositories is very outdated. The current state of rever.xsh is not enough to update them.

sbillinge, 2021-04-15:
@jlhitt1993 rever may be broken following a change in policy at GitHub (permissions lost that rever was using).  I am not sure if it is fixed yet.  we can do manual releases in the absence of rever.  But that doesn't change the concern if code is migrating in the wild but not being pushed to the repo!....

sbillinge, 2025-01-18:
After the repackaging we have a new release workflow that uses GitHub actions.  After this is merged to main I can test it.

We can then work to give release rights to whoever needs it.

sbillinge, 2025-01-18:
closed by #188.  Releases now handled by GH workflows

closed 2025-01-18 by sbillinge