juami/pytentiostat #171: Software JUAMI potentiostat - voltage adjustment problem

Ozturk11, 2021-08-23:
We try to make up a potentiostat system for the CV measurements of our supercapacitor system. Then we installed Labview 2019, Arduino Software and JUAMI.exe files on our local computer and created a circuit for electronic components.

But we had a trouble with Software while adjusting voltage average. We choose a value range as an input  (just as; negative 0, positive 1) but it does not exist. We always read the same voltage value 2.5 Volt in the graph.

May you let us solve this problem.

Ozturk11, 2021-08-23:
We enter the potential value, but it does not reflect the range we entered in the chart

jlhitt1993, 2021-08-24:
This website is a repository for the python version of the software for the JUAMI potentiostat. Feel free to download the python code and give it a try. If you would like help for the LabVIEW version, just send me an email jlhitt1993@gmail.com