juami/pytentiostat #172: wrong port connected - error

PietroCattaneo, 2022-05-26:
Hi, when i try to start an experiment this error comes out: "wrong port connected". how can i solve it? thanks

[image]

jlhitt1993, 2022-05-27:
@PietroCattaneo A couple of things to try. First, click the little drop down arrow at the top next to COM9 and see if there are any other options to try. If not, double check which COM port your Adruino is connected to using Windows Device Manager. It should be in the section called Ports. If you do not see your Arduino in the Device manager, then you are having a problem with the Arduino driver. I would suggest you to reinstall the drivers from Arduino.

PietroCattaneo, 2022-05-28:
The Arduino COM port, in my case COM12, is the only one avaiable. I tryed to reinstall the drivers of Arduino but it says that the ones already installed are "the best ones". 
Maybe it's a stupid question: our Arduino is connected to the computer via USB port... is that the problem? Or do we need to connect the arduino with the other "COM" port? (the one that interact with screens)

jlhitt1993, 2022-05-28:
Yes, the Arduino should be connected to a USB port. So did you confirm that the Arduino is connected to COM12 with the Windows device manager? If so, is COM12 showing up in the pytentiostat software? I see in your screenshot that you tried using COM9.

PietroCattaneo, 2022-05-28:
Right sorry, in the screenshot is indicated the COM9 port because we have two potentiostats: one with arduino 1 (COM12) and the other one with Arduino NANO (COM9). For both the potentiostats the drivers of arduino are fine, and they are recognized in the port section in device manager, so we don't understand the error. if i connect both the potentistats simultaneusly the two ports 9 and 12 appear correctly and i can choose the one to use for the experiment.
The problem appear when i start the experiment with the button "start experiment".

jlhitt1993, 2022-05-28:
I see. The software has never been tested with two Arduinos connected to the computer at the same time. Could you try again with the Arduino NANO disconnected from the computer? The only other thing I could think of is to try to reupload the firmware to the potentiostat found here https://github.com/juami/potentiostat_firmata.

PietroCattaneo, 2022-05-29:
I tryed to reapload the firmware to the potentiostat and now it's working correctly! both the arduino UNO and NANO are working without errors. 
Thank you so much for your time and advices.

jlhitt1993, 2022-05-29:
Glad it worked out. Let us know if you have any other questions or suggestions and thanks for using the potentiostat!

p-0tentiostathuman, 2023-02-05:
PietroCattaneo, jLhitt1993, all of you save my academic life, thank you very much i love you all of you even if i don´t know who the hell are you, even if we never met in person. Again THANK YOUVERY MUCH

sbillinge, 2023-02-08:
> PietroCattaneo, jLhitt1993, all of you save my academic life, thank you very much i love you all of you even if i don´t know who the hell are you, even if we never met in person. Again THANK YOUVERY MUCH

Glad you found this helpful @p-0tentiostathuman