juami/pytentiostat #173: Circuit error

PietroCattaneo, 2022-06-13:
Hi, I'm Pietro, the italian student of Politecnico of Milan. Me and my group are currently working at our bachelor thesis and we finally manage to have a working potentiostat following this project.
I’m writing to you because we believe there is an important error on your first article: "An Easily Fabricated Low-Cost Potentiostat Coupled with User-Friendly Software for Introducing Students to Electrochemical Reactions and Electroanalytical Techniques" of 2018, that is the first article we used to start the project, and then we changed the softwere from LabVIEW to Python.
Here there is the file I'm referring to:
[supporting information for JUAMI.pdf](https://github.com/juami/pytentiostat/files/8892091/supporting.information.for.JUAMI.pdf)

The error I'm talking about is in the drawing of the circuit (figure S6): the node before the reference electrode (RE), that connect the 1K and 10K resintance, is wrong. Intead of this node there should be a jump, so that the two circuit's lines have no connection.

Once we found the error in the schematic and changed the connections of our circuit the potentiostat started to work properly. 
Please let me know if you find the same mistake on your hardwere after checking the circuit connections. 
Thank you, Pietro.

julianaperez, 2022-06-14:
Thank you for answering me, ¿Can you send me the new scheme? I need this
because I cant finish my project.

Thanks

Juliana Pérez

jlhitt1993, 2022-06-14:
@PietroCattaneo Yes, you right. We recently discovered this figure does not match the real circuit in the file for the PCB board. I'm glad you were able to figure out the solution.

jlhitt1993, 2022-06-14:
@julianaperez you can see the correct circuit by looking at the photo in figure S4. Take a look at the connection to Ref. It connects to the op amp only and is not connected to any other wires.