juami/pytentiostat #199: fix: Cannot open port

Dbardgett, 2025-01-21:
### Problem

Attempting to connect to my potentiostat via the "Find Potentiostat" button returns the error "Error. Could not open COM port" and crashes the software. Background: My computer is an M2 Macbook pro which uses a USB <-> USB-C converter to connect, so that could be part of the issue.

### Proposed solution

The error message appears to come from a line in src/pytentiostat/routines.py, but I am not sure exactly what the problem is.

aplymill7, 2025-02-07:
I think this is due to how we were detecting the COM ports. We originally hardcoded the common windows command to list the usb ports. PR #133 was intended to incorporate a specific linux OS port list. 

I feel like the best general practice would be to make a "List COM port method" and we switch case based on OS and then cover some common ones like Windows 10/11, Latest Mac, Ubuntu, Debian, and maybe a Raspberry Pi OS if we ever get a good serial COM case working there

sbillinge, 2025-02-07:
If there are different options we can also wrap things in `try/except` clauses, so it just keeps trying all the options until something works?